kercre123/wire-pod #183: Unable to set Username by 'My name is' command

jstech84, 2023-09-08:
I've recently gotten the wire-pod server up and running on a Raspberry PI 4, and my Vector was successfully connected, and responding to commands. I'm able to use the local host to manage the configuration as well. 

However; I am unable to create a face with the 'My name is' command. I've checked the journal and logs and see that the command is being registered, but the Vector does not process anything and seems to time out. 

The Vector Version is 2.0.1.6076ep

Nismonx, 2023-09-09:
Have you tried asking:
 'who Am I' ?
Should return the answer:
'I don't know'

It toke me 2 attenmpts to get it to register my face. In the end I've presed the buttun, waited a secund and stated: 
'My name is xxx '

Its a standard Vector feature and even with the DDL app and serveres it wasn't alway succesfull with the first atempt. allway had trouble reconiging more that one face. 
Try ussing Morrow Mode or Camera ON to confirm Vector does have clear view, some times too much glare or not enouth light can cause issues.

jstech84, 2023-09-10:
Hi, Yes, I've tried all of these. I've done this at least 20 times and it doesn't complete the process. I can see in the logs that it recognizes the command, but it does not do anything. The lights flash on the unit as it processes, then it drops its head and does nothing. 

Is there by chance a Debug mode on the server to see more verbose logging?

jstech84, 2023-09-13:
After 40 attempts or so, it finally worked. Closing issue.

closed 2023-09-13 by jstech84