kercre123/wire-pod #185: wire-pod setup failure

Toshlaptop, 2023-09-09:
Hi,
I have updated Vector OK but when setting up wire-pod on my Raspberry Pi (Model 3 B) i get the following error followed by loads of stuff.
pi@raspberrypi:~/wire-pod $ sudo ./chipper/start.sh
Initializing variables
SDK info path: /home/pi/.anki_vector/
API config successfully read
Loaded 54 intents and 54 matches (language: en-US)
Initiating leopard voice processor with language en-US
Initializing 3 Picovoice Instances...
Initializing Picovoice Instance 0
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x0]

Any ideas?

F4310, 2023-09-09:
Hello.
Try another OS, in my case last release of Linux Armbian dont work, só a tried a old version and It Works well.

Toshlaptop, 2023-09-09:
> 

Thanks will give it a try

closed 2023-09-24 by kercre123