kercre123/wire-pod #188: Wondering how to configure openai string and model

MitchllB, 2023-09-17:
I was looking to adjust the open ai string prompt, and change the model in the wirepod-chipper-pkgs-knowledge.go file. But I seem to be having trouble executing the build, and seeing these changes. I’m quite a novice when it comes to working with code, so I’m really not sure why I don’t see changes.

mrchristopherstewart, 2023-09-22:
When you get to step 3 on the installation guide (Clone the directory with the following commands.), are you cloning your forked repo (instead of the original without your changes) and installing your server from that? 

As far as I'm aware, that is the only change during installation that would need to be made. I will be trying this out soon though as I would like to change the model and API endpoint due to the currently used ones becoming depreciated. I will update if I find more!

MitchllB, 2023-09-22:
Hey Thank you! yes that's exactly what I ended up doing! just cloned from my changed repo.. I changed the text prompt, then tried to change the model, but evidently I missed some key details. If you get turbo 3.5 or any other model implemented I'd love to see what I did wrong, for now glad davinci is being supported with turbo.

closed 2024-05-28 by kercre123