kercre123/wire-pod #191: Vector not reponse "Hey Vector"

nguyenhunga5, 2023-09-23:
My vector cannot response "Hey Vector" command and after pressing the back button, I say something but do not get any response.
I checked on log, I saw some information
[image]

[image]

Thank you for doing great.

bliteknight, 2023-09-24:
You may have bad microphones. Watch this video to get to the hidden menu and see what the values are for the mics
https://www.youtube.com/watch?v=hOt4iw17yn0

nguyenhunga5, 2023-09-24:
> You may have bad microphones. Watch this video to get to the hidden menu and see what the values are for the mics https://www.youtube.com/watch?v=hOt4iw17yn0

Thank you for your response, I cannot go to hide the menu, I don't know the reason. But yesterday, I went go hide the menu and saw the microphone work.

nguyenhunga5, 2023-09-24:
> You may have bad microphones. Watch this video to get to the hidden menu and see what the values are for the mics https://www.youtube.com/watch?v=hOt4iw17yn0

after teardown my vector and cleaning the cable, it working now. Thank you so much.

closed 2023-09-24 by nguyenhunga5