kercre123/wire-pod #193: Memory leak - back again

bliteknight, 2023-09-24:
So I Implemented an public server that people could connect to for voice processing - 2CPU 4GB Server
And over time the memory usage creeps up and up till its close to 98%
Sitting at 785M in the image below, but it will continue to grow over time

[image]

I initially thought maybe it was all the debug information, so I set debug to false in ~/wire-pod/chipper/source.sh
```
export DEBUG_LOGGING=false
export STT_SERVICE=vosk
``` 
but it is still growing. I have chipper set to run as a daemon, and as soon as I restart the service the memory drops down to < 200mb

[image]

Not sure what to look at that to find the culprit. 

Could the calls to Vectors be causing a memory leak because the server can't ping them?

kercre123, 2023-09-24:
Try adding `JDOCS_PINGER_ENABLED=false` to the source.sh file. I added that specifically for if anyone wanted to run wire-pod on a public server, but I guess I never added it to any documentation

kercre123, 2023-11-06:
Should be fixed now

closed 2023-11-06 by kercre123