kercre123/wire-pod #198: escapepod.local works fine, but vector still cannot connect

KenwoodFox, 2023-10-03:
[image]

Like what it said! I was able to get the debian host put together pretty easy, I'm running on a VM in my homelab, everything seems to resolve fine! But.. no matter what I do, i cant seem to get vector to connect or see it.

[image]

My bot is running version `v2.0.1.6076 0dd1b62e` and I'm using the latest commit from the github, Is it ok if i post the logs downloaded from the bot? Let me know I'll attach them to this issue! And thank you!

KenwoodFox, 2023-10-03:
[image]

KenwoodFox, 2023-10-03:
Hey, that sure looks like a connection in the server journal?...
```
Oct 02 23:07:34 escapepod start.sh[408]: Token expires: 2023-10-03T23:07:34.860927094-04:00
Oct 02 23:07:34 escapepod start.sh[408]: ESN not found in store or this is an associate primary user request, act as if this is a new robot
Oct 02 23:07:34 escapepod start.sh[408]: Adding 10.85.5.211 to TokenHashStore
Oct 02 23:07:34 escapepod start.sh[408]: UUID for this token request: 0c877cdc-783a-4d3a-b137-178b3a70fd19
```

10.85.5.211 is the robot dhcp, checked with my router.

So.. whats it doing?

KenwoodFox, 2023-10-03:
Dosn't seem like it registered properly with the escapepod server, since i cant select it from the dropdown 
[image]

uandco, 2023-10-04:
What do you have in `wire-pod/chipper/jdocs/botSdkInfo.json` ?

KenwoodFox, 2023-10-04:
> What do you have in `wire-pod/chipper/jdocs/botSdkInfo.json` ?

```
# cat wire-pod/chipper/jdocs/botSdkInfo.json 
{"global_guid":"tni1TRsTRTaNSapjo0Y+Sw==","robots":[{"esn":"0dd1b62e","ip_address":"10.85.5.211","guid":"WCbjhAKnnm1gmr5TXr99PA==","activated":true}]}
```

uandco, 2023-10-04:
OK so it's definitely activated, what about `jdocs.json` ?

KenwoodFox, 2023-10-04:
> OK so it's definitely activated, what about `jdocs.json` ?

Here you go!

```
# cat wire-pod/chipper/jdocs/jdocs.json 
[{"thing":"vic:0dd1b62e","name":"vic.AppTokens","jdoc":{"doc_version":1,"fmt_version":1,"client_metadata":"wirepod-new-token","json_doc":"{\"client_tokens\":[{\"hash\":\"Boi3HrTPfUxS1dMAuXR7R92EN5Idh4NtKx6+PKuJKYwvz7j5kljfnJdD6meNaj6w\",\"client_name\":\"wirepod\",\"app_id\":\"SDK\",\"issued_at\":\"2023-10-04T13:34:58.128468087-04:00\"}]}"}},{"thing":"vic:0dd1b62e","name":"vic.UserEntitlements","jdoc":{"fmt_version":1,"json_doc":"{\n   \"KICKSTARTER_EYES\" : false\n}\n"}},{"thing":"vic:0dd1b62e","name":"vic.RobotSettings","jdoc":{"fmt_version":2,"json_doc":"{\n   \"button_wakeword\" : 0,\n   \"clock_24_hour\" : false,\n   \"custom_eye_color\" : {\n      \"enabled\" : false,\n      \"hue\" : 0,\n      \"saturation\" : 0\n   },\n   \"default_location\" : \"San Francisco, California, United States\",\n   \"dist_is_metric\" : false,\n   \"eye_color\" : 0,\n   \"locale\" : \"en-US\",\n   \"master_volume\" : 4,\n   \"temp_is_fahrenheit\" : true,\n   \"time_zone\" : \"America/Los_Angeles\"\n}\n"}}]
```

What am i looking at?

uandco, 2023-10-04:
This is the 2 config files wire-pod uses to communicate with Vector and stores its current config (eye colour etc, what you're supposed to access via the UI). `0dd1b62e` appears in both so it's puzzling, it has everything it needs to communicate.

Can you confirm you're now on the `ep` firmware? Seems like you were not in the other thread.

NonaSuomy, 2023-10-08:
Can you check ps aux and see if avahi-daemon is running and can you check netstat -ltnp and see if you see avahi-daemon running on port 5353 udp

Hamlet3000, 2023-11-07:
I have a similar problem with the last step "Activate"

https://escapepod.local/ shows 404 page not found
http://escapepod.local/ shows the Apache2 Debian Default Page
http://escapepod.local:8080 redirects me correctly to the wire-pod interface.

`pi@escapepod:~ $ service avahi-daemon status
● avahi-daemon.service - Avahi mDNS/DNS-SD Stack
     Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-11-07 10:34:34 CET; 11min ago
TriggeredBy: ● avahi-daemon.socket
   Main PID: 400 (avahi-daemon)
     Status: "avahi-daemon 0.8 starting up."
      Tasks: 2 (limit: 3933)
        CPU: 452ms
     CGroup: /system.slice/avahi-daemon.service
             ├─400 avahi-daemon: running [escapepod.local]
             └─414 avahi-daemon: chroot helper`

`pi@escapepod:~ $ sudo netstat -ltnp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      591/sshd: /usr/sbin
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      550/cupsd
tcp6       0      0 :::8080                 :::*                    LISTEN      1428/./chipper
tcp6       0      0 :::443                  :::*                    LISTEN      1428/./chipper
tcp6       0      0 :::80                   :::*                    LISTEN      622/apache2
tcp6       0      0 :::22                   :::*                    LISTEN      591/sshd: /usr/sbin
tcp6       0      0 ::1:631                 :::*                    LISTEN      550/cupsd`

I'm on the latest wire-pod build

F4310, 2023-11-07:
Its only a suggetion. May IPv6 is causing this problema?

closed 2024-05-28 by kercre123