kercre123/wire-pod #207: Segfault running example under plugins

NonaSuomy, 2023-10-12:
```bash
Bot 1 Intent Sent: intent_system_noaudio
No Parameters Sent
Bot 1 Stream type: OPUS
(Bot 1, Vosk) Processing...
(Bot 1) End of speech detected.
Bot 1 Transcribed text: hello world
Text matched plugin SDK Plugin Test, executing function
hello world plugin test
2023/10/12 13:57:03 vector-go-sdk error: Error opening chipper/jdocs/botSdkInfo.json, likely doesn't exist
2023/10/12 13:57:03 open chipper/jdocs/botSdkInfo.json: no such file or directory
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x7f58a979b16f]
goroutine 125 [running]:
github.com/fforchino/vector-go-sdk/pkg/vector.(*Vector).BehaviorControl(0x0?, {0x1330c90?, 0xc00014a040?}, 0x132ac20?, 0xc00054efd0?)
	/root/go/pkg/mod/github.com/fforchino/vector-go-sdk@v0.0.0-20221216080054-dfe1a258d335/pkg/vector/behaviorcontrol.go:13 +0x2f
plugin/unnamed-5c8a7c9cd58ab1d39d4dbbbd73072c17e0e78bec.Action.func1()
	/wire-pod/chipper/plugins/sdkTest.go:23 +0x33
created by plugin/unnamed-5c8a7c9cd58ab1d39d4dbbbd73072c17e0e78bec.Action
	/wire-pod/chipper/plugins/sdkTest.go:22 +0x12a
exit status 2
```

rwscarb, 2023-11-09:
Was running into the same issue. Turned out I needed to define an environment variable: `WIREPOD_HOME` that is picked up by [fforchino/vector-go-sdk](https://github.com/fforchino/vector-go-sdk/blob/main/pkg/vector/vector.go#L88)

rwscarb, 2023-11-13:
> Can you show me your full test code sample? Still a little confused about how to do that.

Sure, this is how I'm running it in Linux

```console
# cd wire-pod
# WIREPOD_HOME=$(pwd) ./chipper/start.sh
```

Or if in Windows, I believe you set them in System Settings.

NonaSuomy, 2024-01-15:
In the docker console from the directory "/wire-pod" I tried "export WIREPOD_HOME=$(pwd) still failed do you have to restart the start.sh after setting it? (I can't do that as the docker will stop if you kill any tasks on it)

NonaSuomy, 2024-01-15:
So this seems fine in newer releases after recompiling docker.

closed 2024-01-15 by NonaSuomy