kercre123/wire-pod #210: Cannot bring cube

CarbonNeuron, 2023-10-21:
When saying "bring me your cube" to vector, he will pick it up and put it away, not bring it to you. This is caused by the server sending him to pick up his cube instead of bringing it. I'm unsure if this is something difficult to add. On production servers you'd be able to ask him to bring me your cube and he'd look for you, go get the cube and bring it to you. 

The server is definitely hearing me say "bring me your cube" so it's not a stt issue.

F4310, 2023-10-23:
Hello. There is no command "bring me your clube", s Vector do an action close to The asked.
Take your cube
Fetch your cube

CarbonNeuron, 2023-10-24:
On the official servers he would fetch the cube, on this server he just picks it up

closed 2024-05-28 by kercre123