kercre123/wire-pod #211: ep Firmware Not Installing(?)

littlequail, 2023-10-24:
I've attempted to factory reset my Vector and install the firmware from https://keriganc.com/vector-wirepod-setup/html/main.html multiple times and it keeps updating it to the v2.0.1.6076 firmware, not the version ending in `ep`. I've set up the wire pod just fine since escapepod.local works perfectly.

I'm running this on my Windows 11 using Ubuntu, WSL, and Powershell but have also attempted to install the firmware from my Android phone to see if that would change anything.

The strange thing is, the firmware version looks incorrect on the wirepod setup page, but correct on his own screen, see attached photos. Any suggestions? I bought this little guy in college and miss him :(
[image]
[image]

Pittiplatschn, 2023-10-24:
I have exactly the same problem. Vector can't connect to the escapepod after firmware-update. Vectors display shows the ep-Version and in my browser the ep is missing. I use pi-hole as DNS Server, so escapepod.local does resolve correctly to my escapepod-IP on all devices in my network.

Pittiplatschn, 2023-10-26:
I managed to activate him now. My first attempt was running wire-pod under WSL Ubuntu (with Windows 11). I think there is still something blocking Vector from connecting to it.
So I tried it with running wire-pod on my Raspberry Pi 3. That worked finally! Also, I used https://vector.techshop82.com/html/main.html for flashing and activation, but I think that didn't matter. I tried that with Ubuntu before as well.

closed 2024-05-28 by kercre123