kercre123/wire-pod #213: Vector and wire-pod in different subnets

volchenok1983, 2023-10-26:
The Ubuntu 22.01 virtual machine is configured on the 192.168.10.0/24 subnet (192.168.10.183). Installed wire-pod according to the instructions.
The vector is connected to Wi-Fi on the 192.168.9.0/24 subnet (192.168.9.89).

The routes are configured and the pings are working:
```
escapepod:~$ ping 192.168.9.89
PING 192.168.9.89 (192.168.9.89) 56(84) bytes of data.
64 bytes from 192.168.9.89: icmp_seq=1 ttl=63 time=111 ms
64 bytes from 192.168.9.89: icmp_seq=2 ttl=63 time=43.0 ms
64 bytes from 192.168.9.89: icmp_seq=3 ttl=63 time=3.95 ms
64 bytes from 192.168.9.89: icmp_seq=4 ttl=63 time=197 ms
64 bytes from 192.168.9.89: icmp_seq=5 ttl=63 time=436 ms
```

Ping and trace from a laptop on a Wi-Fi network:
```
ping escapepod.local
Exchange packets with escapepod.local [192.168.10.183] with 32 bytes of data:
Reply from 192.168.10.183: number of bytes=32 time=4ms TTL=63
Reply from 192.168.10.183: number of bytes=32 time=1ms TTL=63
Reply from 192.168.10.183: number of bytes=32 time=1ms TTL=63
Reply from 192.168.10.183: number of bytes=32 time=8ms TTL=63
```

```
tracert escapepod.local

Trace route to escapepod.local [192.168.10.183]
with a maximum number of jumps of 30:

   1 7 ms 2 ms 1 ms 192.168.9.1
   2 1 ms 5 ms 5 ms escapepod.local [192.168.10.183]

Tracing is complete.
```

At the same time, I get an error in the web interface:
_Error logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again._

**If Vector and wire-pod are on the same network, then everything works!**

What am I doing wrong?

F4310, 2023-10-26:
Vector looks for a hostname and not for an IP address. Make sure that the virtual machine is configured with hostname "escapepod"
After, ping escapepod.local

volchenok1983, 2023-10-26:
> Vector looks for a hostname and not for an IP address. Make sure that the virtual machine is configured with hostname "escapepod" After, ping escapepod.local

Thanks for the answer.
The machine is configured with the name "escapepod"
Everything is fine with ping.

```
>ping escapepod.local

Exchange packets with escapepod.local [192.168.10.183] with 32 bytes of data:
Reply from 192.168.10.183: number of bytes=32 time=1ms TTL=63
Reply from 192.168.10.183: number of bytes=32 time=39ms TTL=63
Reply from 192.168.10.183: number of bytes=32 time=267ms TTL=63
Reply from 192.168.10.183: number of bytes=32 time=1ms TTL=63

Ping statistics for 192.168.10.183:
     Packets: sent = 4, received = 4, lost = 0
     (0% loss)
Approximate round trip time in ms:
     Minimum = 1ms, Maximum = 267ms, Average = 77ms
```

Perhaps it's an avahi issue, since this service only works within one subnet?

kercre123, 2023-11-06:
It could be a problem with Vector's old avahi. I've never tried it with multiple subnets

closed 2024-05-28 by kercre123