kercre123/wire-pod #214: Issue with Bot Settings

apeoplerson, 2023-10-27:
Not sure why this is happening. Hopefully it something easy.

error: rpc error: code = Unimplemented desc = unknown service Anki.Vector.external_interface.ExternalInterface

elliot-holley, 2023-11-01:
Having the same issue, running on a vector 1.0 on WSL. everything else works fine.

bumbleb0321, 2023-11-07:
I believe I am also having the same issue through wsl on a Windows 11 device. When trying to open bot settings (to set up voice among other things) I receive the error pop up :

escapepod.local:8080 says
error: rpc error: code = Unimplemented desc = unknown service
Aniki.Vector.external_interface.ExternalInterface

bliteknight, 2023-11-13:
You cannot connect to your bots when you install via WSL

Wire just released a native version that runs on windows so you can give that a try

closed 2024-05-28 by kercre123

lukeb12798, 2025-02-18:
I just used the windows version but it said the same thing.