kercre123/wire-pod #215: issue with continuing installation of wirepod on a raspberry pi.

SgtKiLLx, 2023-11-01:
upon attempting to install, it appears its not downloading GO. ive installed it manually, and still getting the same issue.

[image]
[image]

F4310, 2023-11-01:
What is the operating system?

F4310, 2023-11-01:
Try using Armbian 23.02.2 Bullseye

SgtKiLLx, 2023-11-02:
> What is the operating system version?

Latest Debian

kercre123, 2023-11-02:
Make sure you're using a 64-bit Raspberry Pi OS. The default Raspberry Pi OS has a 64-bit kernel, but links with 32-bit, and setup.sh can't detect that at the moment.

SgtKiLLx, 2023-11-05:
Will try.

SgtKiLLx, 2023-11-07:
alright. chipper running. as per the following step(s)

[Authenticate a production bot]

It is recommended to clear your bot's user data. This is not required, and you can still authenticate with wire-pod without it (as long as the last server you have authenticated the bot with was the DDL/Anki production stack), but it may cause unexpected behavior.

Place Vector on the charger
Double press his button
Lift his lift up then down
Take Vector off of the charger and twist one of the wheels until the cursor is on "RESET" (or "CLEAR USER DATA")
Lift the lift up then down again
Move the wheel until the cursor is on "CONFIRM"
Lift the lift up then down again

**Refresh** the [vector-wirepod-setup](https://keriganc.com/vector-wirepod-setup) page and follow the instructions  <<<<<-----  I get to here, refreshing does nothing. 

[image]

with V all brand new, with a new ID#, hes jus sittin on his charger with his head up with anki.com/v across his face

kercre123, 2023-11-07:
Follow the instructions on the page. It will detect that Vector isn't in recovery mode and will authenticate him rather than put a different firmware on him.

jowensfog, 2023-12-05:
Just wanted to say I ran into this same issue and following the instructions, was able to get it resolved. I have a brand new raspberry pi 4b that came with a pre-installed OS. I ended up downloading the latest 64bit os and started from scratch and that helped.
And then during the setup stage I also got Vector into the state described above. I realized I needed to double-tap Vector's button to get into pair mode. And then after updating the firmware when that Pair With Vector button was shown, to then just double-tap Vector's button again. On the first go around I made the mistake of holding his button down for 15 seconds to pair him again (which I think reset his firmware). That may have been obvious to others, but thought I'd say that here in case anyone else new makes that mistake.

closed 2024-05-28 by kercre123