kercre123/wire-pod #216: Can't setup vector with the escapepod firmware.

TheGithubJoshua, 2023-11-02:
On the update screen the progress bar never moves, I have waited hours any help?
[image]

seamonkey420, 2023-11-03:
just wanted to chime in since i was trying to get my wire-pod setup going today and yesterday.  i also was having major issues getting the escape pod firmware to download, mine would finally start downloading but after about 20 mins would stop and my vector would have a flashing ! on the screen. 

i then restarted my vector, put back into recovery mode, repaired and tried to download again. simiar issue after another 30 mins of waiting.  gave up for the night last night.  

today i decided to reset user data on my vector, then put back into recovery mode and paired to laptop. was able to finally get firmware downloaded but it did give me a ! twice so i had to restart but it seems like files continued from prev spot.  then tried to activate and had issues with not able to find server.  checked server, all services were running as they should be.  again put vector in recovery mode, repaired and went to firmware site and it downloaded again without any !s on the screen and went to activation mode. was finally able to activate and setup alexa.

so yea, i think somethings up on the firmware website. i'm actually amazed i got it up and running today with all the issues i had w/getting escape pod firmware on my vector.  thx to kercre123 and everyone for this.

TheGithubJoshua, 2023-11-04:
Solved just by trying again using
[The epod setup page](https://keriganc.com/vector-epod-setup/html/main.html)
So definitely somethings up with that website @kercre123

TheGithubJoshua, 2023-11-05:
Decided to reopen because this is still an issue with the setup page even though the epod setup page works fine.

kercre123, 2023-11-06:
It works fine for me. vector-wirepod-setup and vector-epod-setup are exactly the same except they give Vector different OTAs. The only reason I can think of for this is maybe you went to the site before, when the OTA address pointed to a DDL server, and that cached. So maybe try CTRL+SHIFT+R. I will leave this open as I do need to create a better setup page which allows you to select the firmare you want.

closed 2024-05-28 by kercre123