kercre123/wire-pod #218: cannot get vector to flash firmware

DIRECTVIRUS, 2023-11-03:
[image]

whenever I try to ota-start using either a locally hosted web server for the file or the link in the guide the vector robot never begins the download and therefore cannot connect to wirepod as it cannot flash to the custom firmware

[image]

kercre123, 2023-11-03:
You have to go to recovery mode first by placing Vector on the charger and holding the button for 15 seconds. The firmware must report as v0.9.0.

DIRECTVIRUS, 2023-11-03:
that was it thank you. thought I had him in recovery mode.

closed 2023-11-03 by DIRECTVIRUS