kercre123/wire-pod #219: Can't change Locale in Vector Web App. Keeps reverting back.

wistxxx, 2023-11-05:
Every time I change locale from English (US) to English (GB) in the configuration section of the vector web app, it reverts back to US.

I get the 'Jdocs: Incoming WriteDoc request, Item to write: vic.RobotSettings, Robot ID:' in my ubuntu window, but after using vector for a bit, it seems to revert back to English (US) in the configuration section.

Is there any way I can view/change this locale setting manually? Or any ideas what's causing this?

Any help much appreciated.

Many thanks

kercre123, 2023-11-06:
Do any other locales work? The way to set it is via SDK request, which is what the web app does. You can view it with `cat ~/wire-pod/chipper/jdocs/jdocs.json`

wistxxx, 2023-11-11:
> Do any other locales work? The way to set it is via SDK request, which is what the web app does. You can view it with `cat ~/wire-pod/chipper/jdocs/jdocs.json`

Many thanks for getting back to me. And for pointing me in the right direction. It seems I was running vectorx on top of wirepod, and that's what is resetting the locale each time a custom intent is called. So not really a wirepod issue I guess (ง •_•)ง

I'll close issue. Thank you again!

closed 2023-11-11 by wistxxx