kercre123/wire-pod #223: Docker speech to text service error

Schattenruf, 2023-11-06:
Hi,

I want to run wire-pod in a docker environment because my port 8080 and 80 are blocked by another resource and the data should be stored on my NAS on volume1.
But if I build my docker script in portainer, I get something like this as a result:

`The command '/bin/sh -c sudo STT=vosk /wire-pod/setup.sh' returned a non-zero code: 1`

This regards in the STT=vosk command not being taken into consideration. It ends with:
`Which speech-to-text service would you like to use?
1: Coqui (local, no usage collection, less accurate, a little slower)

2: Picovoice Leopard (local, usage collected, accurate, account signup required)

3: VOSK (local, accurate, multilanguage, fast, recommended)

`
How can I skip that?

Kind regards,
Schattenruf :)

Docker build taken from the 1st post here: https://github.com/kercre123/wire-pod/issues/201#issue-1929594044

kercre123, 2023-11-06:
The script should be reading that STT=vosk variable and skipping the prompt... maybe try adding `ENV STT=vosk` in the line before `RUN sudo STT=vosk /wire-pod/setup.sh` ?

Schattenruf, 2023-11-06:
Hi,

Thanks for your reply. I tried to install it directly on my pi with the
STT=vosk variable set, but it skips it as well (the issue seems unrelated
to docker).

Maybe because of the setup of the script itself. It asks for your input and
after that checks if you set a variable.

Schattenruf, 2023-11-07:
Edit: The ENV did not help either :(

Schattenruf, 2023-11-07:
I solved this problem by running
`RUN echo "3" | sudo STT=vosk /wire-pod/setup.sh
`

kercre123, 2023-11-07:
For me, running `sudo STT=vosk ./setup.sh` on a normal x86_64 computer results in the vosk config being used without any prompting. The script looks at the STT variable, and if it isn't `vosk`, it will prompt the user. The STT prompt function is defined before that happens, which is why it may seem that code happens before the check

Schattenruf, 2023-11-08:
>
> Hi then this is really strange. I loaded your files on a brand new pi with
> raspbian. There the vosk variable is not going to be used.
>
It awaits an input.

> Message ID: ***@***.***>
>

kercre123, 2023-11-09:
Is it regular 32-bit Raspberry Pi OS or 64-bit? If it detects 32-bit armv7l, the STT variable gets unset and it should tell you to use Picovoice.

Schattenruf, 2023-11-09:
> Is it regular 32-bit Raspberry Pi OS or 64-bit? If it detects 32-bit armv7l, the STT variable gets unset and it should tell you to use Picovoice.

If I run it in docker it's Ubuntu. But yes host is 32bit armv71.
But I got it working it seems with VectorX installation :)

closed 2023-11-09 by Schattenruf