kercre123/wire-pod #225: Wire-pod server crash 9 novembre 2023

krunch-krunchy, 2023-11-10:
Hi, i've recently installed wire-pod on win10, but when i'm trying to start the wire-pod server it crashes;
It gives me thoses errors and i don't how i could solve them:

 [zeroconf] no suitable IPv4 interface: udp4: failed to join any of these interfaces: [{7 1500 eth2 8b:c6:3b:1d:6a:59 up|broadcast|multicast} {61 1500 eth3 0b:00:27:00:00:3d up|broadcast|multicast}]
wire-pod started successfully!
 [zeroconf] no suitable IPv6 interface: udp6: failed to join any of these interfaces: [{7 1500 eth2 8b:c6:3b:1d:6a:59 up|broadcast|multicast} { up|broadcast|multicast}] 
exit status 2

Thanks in advance

kercre123, 2023-11-10:
The latest commit should fix this. Run:

```
cd ~/wire-pod
git pull
```

Then try starting it again.

krunch-krunchy, 2023-11-10:
Thanks, it solved this problem, now i'm habing trouble pairing the vector setup page with vector but it's another issue, thanks for your help

closed 2024-05-28 by kercre123