kercre123/wire-pod #233: change vector’s wifi

Pane82, 2023-11-20:
there is a way to change vector’s connected wifi without wiping his data?

edd134, 2023-11-20:
I'm pretty sure if you start wire pod this new wifi network, you can choose this new one on vector setup (not sure tho)

Pane82, 2023-11-20:
i’ve tried but he cannot connect to the pod…

edd134, 2023-11-20:
turn off the old wifi

Pane82, 2023-11-20:
well he auto select it anyways

edd134, 2023-11-20:
are you  going onto the vector setup page?

Pane82, 2023-11-20:
yep i’ve tried but it set the old wifi and i cannot change it

kercre123, 2023-11-20:
Theoretically, you should be able to head to [noflow-devsetup](https://www.project-victor.org/noflow-devsetup) and use BLE commands to get him to forget a network and join a new one. Connect like normal, but do NOT check "Enable auto-setup flow" when connecting. Use `help` to figure out what commands do what. For `anki-auth`, you should be able to use any random string (ex. `anki-auth asdjoasdoiqwoijd`).

bliteknight, 2023-11-23:
> Theoretically, you should be able to head to [noflow-devsetup](https://www.project-victor.org/noflow-devsetup) and use BLE commands to get him to forget a network and join a new one. Connect like [...]

This is the only way I have verified to change his Wifi. Once you get in the "noflow" terminal, there is a command to wipe all wifi connections - but you have to anki-auth before you do so. Once you run that command you will be able to clear all connections and join the new one using 
wifi-scan
wifi-join "Wifi Access point" "Wifi password"

Hitojoy, 2024-01-17:
> Theoretically, you should be able to head to [noflow-devsetup](https://www.project-victor.org/noflow-devsetup) and use BLE commands to get him to forget a network and join a new one. Connect like [...]

anki-auth won't work,failed to authorize

Zomboy4313, 2024-01-17:
What I do is put vector on his charger, double press his back button and connect. Then I just go though the setup process, and connect to new WiFi. Did this to my phone hotspot earlier today.

closed 2024-05-28 by kercre123