kercre123/wire-pod #234: The new wire pod native windows app is broken

edd134, 2023-11-20:
So I was using the wsl version of wire pod and it worked fine, vector would connect and respond to voice commands. And then I got the native windows version thinking it would solve my bot settings issue relating to wsl, it started fine and I set my PC name to "escapepod" but when I press the activate button on the vector setup page I get the message "Error logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again."
[image]

Pane82, 2023-11-20:
try to see if the firewall is blocking connection

edd134, 2023-11-20:
[image]
nope it isnt

edd134, 2023-11-20:
i dunno if this is useful to fix it but if i go into system tray and click web interface, it doesnt work so i have to type localhost:8080 instead

Pane82, 2023-11-20:
no it doesn’t fix because vector needs to connect to escapepod.local 
are the wirepod and vector on the same wifi?

edd134, 2023-11-20:
vector and wire pod are on the same ip, but the whole escapepod.local is a mystery to me

Pane82, 2023-11-20:
try to ping it from another device on the same wifi so you can see if escapepod.local works

edd134, 2023-11-20:
when i do it on mobile it straight up just wont connect vector to wifi even tho i have double checked that it is the right one

Pane82, 2023-11-20:
try to ping the escapepod.local if it doesn’t get a ping the domain dosn’t work

Pane82, 2023-11-20:
ping escapepod.local

edd134, 2023-11-20:
Pinging escapepod [fe80::7969:1749:2499:d7f6%4] with 32 bytes of data:
Reply from fe80::7969:1749:2499:d7f6%4: time<1ms
Reply from fe80::7969:1749:2499:d7f6%4: time<1ms
Reply from fe80::7969:1749:2499:d7f6%4: time<1ms
Reply from fe80::7969:1749:2499:d7f6%4: time<1ms

Ping statistics for fe80::7969:1749:2499:d7f6%4:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 0ms, Maximum = 0ms, Average = 0ms

Pane82, 2023-11-20:
escapepod.local work, do you have upnp?

edd134, 2023-11-20:
im pretty sure i dont but ill double check

edd134, 2023-11-20:
activated now

edd134, 2023-11-20:
its not lan connected

Pane82, 2023-11-20:
try now

Pane82, 2023-11-20:
try now

TheGithubJoshua, 2023-11-20:
Have you changed your hostname to escapepod?

edd134, 2023-11-20:
yep

Pane82, 2023-11-20:
vector updated in the first pair?

edd134, 2023-11-20:
wdym by first pair

Pane82, 2023-11-20:
when it gives the error

edd134, 2023-11-20:
yes its updated

edd134, 2023-11-20:
UPDATE: when i type ping escapepod.local in the linux terminal, it says "64 bytes from escapepod (172.20.109.102): icmp_seq=40 ttl=64 time=0.021 ms" repeatedly

edd134, 2023-11-20:
i reinstalled it and it still isnt working

Pane82, 2023-11-20:
Double check that your windows pc name is escapepod

edd134, 2023-11-20:
it is

Pane82, 2023-11-20:
Well now I really don’t know how to help you, I will search online later.

kercre123, 2023-11-20:
This is not a messaging platform. People (like me) are subscribed to this repository, and we don't want our emails spammed. Think of suggestions in batches, and answer in batches. For a more message-like experience, join the [Discord](https://discord.gg/NjcAjZANFS) and discuss in the #wirepod-help thread.

When you changed the computer's hostname to `escapepod`, did you restart the computer afterwards?
Is there any other device on the network possibly called `escapepod`?
After you activated, did wire-pod crash? (the current release version doesn't show an error message, it just crashes silently. You'll have to check the system tray and see if wire-pod is still there)

closed 2026-11-15 by kercre123