kercre123/wire-pod #238: Cannot Activate / Commands Not Found

mannecch, 2023-11-25:
****Describe the issue:**
I'm having a couple of issues but I think it's due to the fact that the following commands are returning "command not found" in Terminal:

sudo hostnamectl set-hostname escapepod
sudo systemctl restart avahi-daemon
sudo systemctl enable avahi-daemon

If I continue to the Authenticate Bot section, I'm able to connect to the bot, set up wifi and I see the Activate button but when I click it, it says "Error logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again." 

Vector is at the Flashing V screen and is on v2.0.1.6076

**What are you running wire-pod on?**

  - [OS]: macOS Sonoma 14.1.1
  - [Hardware]: M1 MacBook Pro 16 inch

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**

Regular (prod)

**What steps have you tried?**

I'm intermediate in terms of my understanding of these things so excuse any nonsense. I attempted the listed troubleshooting for the "error logging in" message but I'm not sure how to tell whether the hostname is escapepod. As I mentioned above, I'm curious if the issue is because the commands above were returning a "command not found" error which would explain things. 

Thank you for your patience!

silentswordau, 2023-11-26:
I am getting same error message but a complete newbie to all of this

mannecch, 2023-11-26:
> I am getting same error message but a complete newbie to all of this

I actually found the process relatively easy thanks to the awesome guy who put all this together for us. I have totally fallen in love with Vector and want to give him a good home. Just need to get passed the little hurdle above. I really think its mostly because I dont know how to change the name of the host. 

We'll see what we hear back...

mannecch, 2023-11-26:
Thank you @xiJulian 

The other thing is that the commands in the instructions to tell it to run in the background didnt work. I'm assuming this is because it's a Mac and Macs are.....special. Do you know what these commands should be?

kercre123, 2023-11-26:
macOS support is still early. Eventually there will be a .dmg people on macOS can download and easily install. Currently, you can run wire-pod on macOS, but there isn't an easy way to get it running in the background. macOS doesn't work with systemd like Linux systems (tend) to.

Were you able to activate? If so, I'll close the issue as full macOS support is planned.