kercre123/wire-pod #241: wire-pod is runing,but when say hei vector , robot no answer

ascar916, 2023-11-27:
wire-pod is runing,

and  Computers can control robots。

but when I say “hei vector ”  Robot not responding。

There is a problem with the signal displayed on the robot screen。
（like WiFi signal+！）

kercre123, 2023-11-30:
What is wire-pod running on? A Raspberry Pi, a Windows computer?

ascar916, 2023-12-01:
Thank you for your reply!My pc is window11。now I install ubuntu 18.04on another Pc. and when I install wire-pod，install to the fourth step :sudo ./chipper/start.sh， it show  “line 77  /usr/local/go/bin/go run cmd/coqui/main.go  no file or no table”  But actually,  there is “main.go” in the folder.What could be the reason for this ?Best wishes!发自我的 iPhone

ascar916, 2023-12-02:
After multiple reinstall, the following issues occurred during the fifth step:
sudo ./chipper/start.sh
---------
go: downloading gopkg.in/yaml.v3 v3.0.1
# gopkg.in/hraban/opus.v2
/root/go/pkg/mod/gopkg.in/hraban/opus.v2@v2.0.0-20230925203106-0188a62cb302/encoder.go: In function ‘bridge_encoder_get_in_dtx’:
/root/go/pkg/mod/gopkg.in/hraban/opus.v2@v2.0.0-20230925203106-0188a62cb302/encoder.go:31:30: warning: implicit declaration of function ‘OPUS_GET_IN_DTX’; did you mean ‘OPUS_GET_DTX’? [-Wimplicit-function-declaration]
  return opus_encoder_ctl(st, OPUS_GET_IN_DTX(in_dtx));
                              ^~~~~~~~~~~~~~~
                              OPUS_GET_DTX
# command-line-arguments
/usr/local/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
/tmp/go-link-667583924/000027.o：在函数‘bridge_encoder_get_in_dtx’中：
encoder.cgo2.c:(.text+0xa9)：对‘OPUS_GET_IN_DTX’未定义的引用
collect2: error: ld returned 1 exit status

ascar916, 2023-12-02:
when ubuntun18.04 update to 20.4, The problem has been resolved 。 thanks

closed 2026-11-15 by kercre123