kercre123/wire-pod #242: Vector Python SDK

ckrokosh, 2023-11-27:
Hi all,

Successfully installed Wire-Pod on a raspberry pi 4 yesterday and all works great. Vector is alive again.

My question is: Can I run the the Vector Python SDK without being able to connect to the DDL servers. I saw mention of it on the Wiki where it states "When a bot is authenticated, the bot is added (or modified if already there) to sdk_config.ini, which allows the Python SDK to continue to work." Does authenticated mean by DDL or wirepod? I have done numerous google searches and haven't found any thing on how to get the SDK working.

Any help on getting the SDK working would be appreciated.

ckrokosh, 2023-11-27:
Thanks. Unfortunately I got a new computer and because i hadn't used Vector for so long i never thought to back up the vector directory...and i don't hold much hope for DDL servers coming back up.

ckrokosh, 2023-11-28:
I went to the link you posted above. Noticed that the anki_vector folder was already on my Raspberry Pi and had the needed files there, as well the sdk_config.ini seemed to be filled out for my Vector including a guid. Wasn't sure if it was a placeholder or not but i installed the Anki Vector SDK via pip. Tried running the example programs and they failed telling me to downgrade "protobuf". After doing that all the examples ran fine so it looks like the SDK runs out of the box after downgrading protobuf.

kercre123, 2023-11-28:
wire-pod automatically configures the .anki_vector folder. To use other computers with the python sdk, you will need to copy the .anki_vector folder for now.

closed 2023-11-28 by kercre123