kercre123/wire-pod #243: [BUG] Cant do any Math

tgwaste, 2023-11-27:
Should "Hey vector, whats 5 plus 5?" work?  It cant seem to do any math at all.

kercre123, 2023-11-27:
If you have a knowledge-graph API selected, then yes. It is not a feature embedded in wire-pod.

kercre123, 2023-11-27:
Houndify is free, but you would have to ask it like this:

"Hey Vector, question"
<wait for Vector to say "Ready!">
"What is 5 times 5?"

OpenAI allows you to skip the "question" part, but it is paid. It is usually less than a dollar a month, depending on usage.

tgwaste, 2023-11-27:
I actually just got an open AI key and its working for free.  I wonder if they have some sort of free limit or something. Im looking around.

kercre123, 2023-11-28:
There is a free trial. Eventually it will start charging. I think the highest charge for one month I got was around $1.50