kercre123/wire-pod #245: Can’t activate vector

Gaudhen, 2023-11-29:
I have wire pod setup on my windows 10 laptop with the app and I’m trying to connect my prod vector to it but it keeps saying error can’t activate bot check your internet connection or something like that I haven’t seen any steps on anything for the firewall or WiFi and I did all of the steps I tried turning off firewall and resetting my router (if this is helpful my router is a NETGEAR C6300) let me know if you can help me because I have been trying to look for a solution for this!

Nitheshwar27, 2023-12-17:
https://www.youtube.com/watch?v=8Hun_F3WGF8 try these steps this should work

Gaudhen, 2023-12-18:
I already tried the steps

Westernllamas, 2024-12-09:
Did you ever get it working? having same problem.

closed 2026-11-15 by kercre123