kercre123/wire-pod #248: mDNS issue on windows server

Schaefchen89, 2023-12-05:
Hey there! Got the problem with my wire-pod... to be specific with the mDNS!

The setup worked fine and everything is running as long as i rename my server to "escapepod"! BUT i need another name for it :D 
So i tried editing the server_config.json from escapepod to the name i need, but that took no effect the whe WIFI ! error... also searched for other solutions, but found nothing :/ Funny thing: I can do everything with the robot (move WASD, ...), but the robot itself is not responding to questions (like weather, ...) as long as the name is not escapepod. So the connection wire-pod -> vector is working fine, but vector -> wire-pod not :( for sure! i could setup a RasPi to do the mDNS management, but thats not needed atm! And if i would setup a RasPi, it would be easier to install the wire-pod directly instead of redirecting with it to the server :D was hoping to run the wire-pod at the existing server

So my question is: is there a way to edit the name, so vector is requesting f.e. "server" instead of "escapepod" as DNS name? Did i miss sth in the wiki? Or do i have to edit sth at the robot? Or may point an static ipv4 (server, devices, ... got static ip's) ?

Wire-Pod is running on a windows server and the prod bot works fine with escapepod, but not with other names.

thx for every input!

kercre123, 2023-12-05:
Vector is hard-coded to communicate with `escapepod.local`. There is no way to change this.

Schaefchen89, 2023-12-05:
Damn! :/ Was afraid of that answer...

But: if i would run an vm-raspi on the windows-server, could that work? Just thinking about this, but it's 4:20am here and i'm braindead :D I'm not sure if the vm-raspi would get listet like an normal raspi in the network with "escapepod" or if there could be an forwarding issue caused by the mDNS of the server?!

closed 2026-11-15 by kercre123