kercre123/wire-pod #251: [HELP] Not recognizing "Hey Vector"

juanmadelboca, 2023-12-13:
**Describe the issue:**
I set up the server without any problem on my windows computer, changed the hostname and everything. After the setup I discover that my vector robot was responding to the "alexa" command without any troubles. And also to the map of knowledge and commands but only if i press the back button.
I tryied a lot of different combinations and for some reason the "hey vector" only worked once after I set up everything after that no matter the tone, the voice or the person (even tried with google translate voice) it never work.
**What are you running wire-pod on?**
  - [OS]: Windows 11
  - [Hardware]: Laptop

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
Is an ANKI one (99% sure is prod one)
**What steps have you tried?**
- Reset vector
- set up wire pod
- power on/off vector
- power on/off server
- set up the server in a MacOS
- power on/off server
- Reset the wire pod serve on windows

(note: I will delete any issues which fit under this category but don't follow this template. I will not accept an "It's just not working!")

kercre123, 2023-12-13:
"Hey Vector" recognition is done in the robot. Wire-pod handles the audio after "Hey Vector". Maybe try different English locales in Bot Settings

towniehunter, 2023-12-13:
Have you activated the robot on the server? I've found it needs to be redone occasionally

bryanajwells, 2024-03-24:
I was having this issue, then I reset the language to English using the UI.  I dont know what language Vector was using but all good now.

bbb5317, 2025-04-06:
> I was having this issue, then I reset the language to English using the UI. I dont know what language Vector was using but all good now.

Are you able to list in more details exactly what you did? There is no such thing as Language. There is Locale. In my Locale it was sayiung English US I set it to English Au - no change. Still not working. Do I need to reset or reboot something after I change language?

bbb5317, 2025-04-06:
> Are you able to list in more details exactly what you did? There is no such thing as Language. There is Locale. In my Locale it was sayiung English US I set it to English Au - no change. Still not [...]

It seems after I set the Locale to English AU and after siting in the charger for about 5 min or so without doing anything else it just started working.

closed 2026-11-15 by kercre123