kercre123/wire-pod #252: [HELP] Cannot activate production Vector

twilsonco, 2023-12-14:
**Describe the issue:**

**What are you running wire-pod on?**
  - [OS]: Tried with macOS 10.15.7, and on Debian 12 (VM)
  - [Hardware]: Intel Mac Mine Late 2012.

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**

Production bot

**What steps have you tried?**

With both the macOS and Debian installs (trying Debian first, then macOS):

* I was able to access the wire-pod interface at http://escapepod.local:8080. 
  * from the wire-pod machine, and from other machines on the network, and from my iPhone
* Was able to pair to the Vector using https://vector.techshop82.com/ in an incognito (or regular) Chrome window using a Mac mini 2018.
* Was able to connect Vector to wifi (image shows networking info on Vector screen)
* Was able to refresh Vector with "ep" firmware (shows 2.0.1.6076ep, 00700ede) at the top-right once paired.
* After pressing "ACTIVATE", I get the message "Error logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again."
  * the vector shows the "V" inside the phone on-screen
  * Downloading logs from the bot, I see in run/update-engine/error, "Failed to open URL: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581)>"
    * My wifi setup is an Apple AirPort Extreme 802.11ac wifi router (broadcasting 2.4 and 5GHz with the same SSID) in bridge mode connected to the actual router. 
    * The Mac running wire-pod has wifi disabled, and is hardwired to the actual router.
  * data/boot.log is attached below
  * Looking at terminal output on the wire-pod Mac machine, nothing is printed during this process; just shows "wire-pod started successfully"
  * ~/wire-pod/chipper/jdocs contains only "placeholder"
  * Everything listed here was precisely the same behavior when trying from the Debian 12 VM.

Thanks in advanced for your work on this. Hope the info here is helpful.

[boot.log](https://github.com/kercre123/wire-pod/files/13668612/boot.log)

[image]

twilsonco, 2023-12-25:
Looking into the vector's logs in data/diagnostics/logs/log.txt, I see that the vector cannot find escapepod.local, even though I can resolve http://escapepod.local:8080 from any other machine on the network. 
```
12-25 21:58:19.965 info vic-switchboard 2159 2159 vic-switchboard: Received cloud session authorization request.
12-25 21:58:20.068 info vic-switchboard 2159 2159 vic-switchboard: Received message from token_server: 4
12-25 21:58:20.068 info vic-switchboard 2159 2159 vic-switchboard: CloudRequest JWT Response Handler
12-25 21:58:20.700 info vic-engine 2138 2138 vic-engine: @action.play_animationanim_lookatphone_loop_01ag_lookatphone_loopOnboardingLookAtPhoneLoopHighStim441562512
12-25 21:58:23.879 info vic-engine 2138 2138 vic-engine: @action.play_animationanim_lookatphone_loop_01ag_lookatphone_loopOnboardingLookAtPhoneLoopHighStim441565692
12-25 21:58:25.085 info logwrapper 2139 2139 vic-cloud: Token queue error: rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp: lookup escapepod.local: no such host"
12-25 21:58:25.085 info logwrapper 2139 2139 vic-cloud: Error handling token request: rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp: lookup escapepod.local: no such host"
```

kercre123, 2023-12-26:
Have you tried the latest commit? I just pushed a bunch of code which should help Vector resolve to escapepod.local. wire-pod, in the latest commit, registers itself as a service rather than relying on the computer's mDNS capabilities

twilsonco, 2023-12-26:
Thanks @kercre123. I just pulled, reinstalled, and restarted the wire-pod machine. It's still accessible to every other device on my network at escapepod.local, but the Vector continues to error and the logs show the same "no such host" in the error. I might try installing wire-pod on a Windows machine to see if that works, but I don't know why it would fix this issue since everything else appears to be working...

twilsonco, 2024-02-02:
Unfortunately, no. I got covered up with work and so I've been letting it sit for a while. This project is very actively maintained, so I was crossing my fingers that one of the updates might fix the issue. Otherwise, at some point I'll just buy a micro PC or an r-pi to see if that helps.

kercre123, 2026-11-15:
stale

closed 2026-11-15 by kercre123

phucbao19910-cpu, 2026-11-15:
same issue, any help?

phucbao19910-cpu, 2026-11-15:
i use a macbook air M4 but no hope connecting to wirepod and it says "Error logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again."