kercre123/wire-pod #255: "sudo ./chipper/start.sh" command producing error [HELP]

BelowAverageGitUser, 2023-12-17:
Issue:
Running "sudo ./chipper/start.sh" works for a few lines, then ends on "../../go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.46/vosk.go:110:2: could not determine kind of name for C.vosk_recognizer_set_grm"
I have followed the steps exaclty, had to install brew on the spot, but other than that not sure if i've somehow messed up.

Running on MacOS Sonoma
Macbook Air M2

Setting up a regular bot

Got to step 5 of guide 1

mdilling, 2023-12-18:
I too am having the same issue.

Running on MacOS Sonoma Version: 14.2 (23C64)
Mac Studio M2 Max

kercre123, 2023-12-18:
macOS building is broken until someone compiles the latest Vosk code to OSX. For now, I recommend using the .dmg release in [https://github.com/kercre123/WirePod/releases/tag/v0.2.0](https://github.com/kercre123/WirePod/releases/tag/v0.2.0)

closed 2026-11-15 by kercre123