kercre123/wire-pod #257: [HELP] Internationalization of voice recognition and speech synthesis

mszczuro, 2023-12-21:
**Describe the issue:**
I don't know if it's issue or am I trying to accomplish the impossible, so I'll ask here.
I want to fully switch Vector to Polish language, so anyone in my household can use it.
My understanding is that Vector reacts to "Hey Vector" *only* and then it sends whatever was spoken afterwards to the cloud / wire-pod.
If I set up locale and voice recognition options to Polish, I should get Polish text as the input for further processing. But does wire-pod understand text in languages other than English?
What about weather/temperature?
What about ML engine? That I understand depend on engine I chose, right?
If commands can and are understood in Polish, do I get answer in Polish and proper TTS afterwards?
I guess I'm asking if anyone managed a full e2e workflow in language other than English.
If yes, have you encountered any problems or you're aware of hacks that need to be made to accomplish that?

**What are you running wire-pod on?**
  - Windows 10 ent x64
  - RPi 4B 8GB

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
PROD

**What steps have you tried?**
RPi OS locale set to PL, console language, charset set to PL, STT in the config set to PL

bc1bb, 2024-01-24:
Hi,

So apparently this issue https://github.com/kercre123/wire-pod/issues/125 fixes how the bot understands and speaks polish. But i am curious at how well it performs in other languages, I am french and wanna know if it would work correctly given our weird prononciation.
I don't think the project is really supported anymore but all your questions are interesting and I wanna hear of the answers.

mszczuro, 2024-01-25:
> Hi,
>
> So apparently this issue #125 fixes how the bot understands and speaks polish. But i am curious at how well it performs in other languages, I am french and wanna know if it would work correctly [...]

Not quite. My bot now understand Polish but speaks English. Even when I ask him what's my name, he use my Polish name but pass it through English TTS.

Maybe I'll find some time on the weekend to go through the wire-pod and bot config to maybe understand how it really works.

kercre123, 2024-01-25:
Vector was only really designed for English. TTS is done on the bot, and the only TTS pack on the bot is English(US). Older OTAs had English, Japanese, German, and Spanish (iirc), but they were taken out in 1.7.

mszczuro, 2024-01-25:
Wow, that's kind of sad.
That also mean my job here is done.
Thank you for clarification!

closed 2024-01-25 by mszczuro