kercre123/wire-pod #263: [HELP] Vector has had issues on multiple set-ups responding to anything after "Hey Vector."

chocchocchip, 2024-01-10:
- The Vector device will react to Hey Vector. **After asking something like "What's my name?" or "Pickup your cube?" the Vector will "think" (the white lights on his head will pulse) and then the query will timeout without yielding a response.**

- The Wirepod Interface looks great. I can see a log and things like:
                  _[date shown] [time shown] Intent matched: intent_play_blackjack, transcribed text: 'play blackjack', device: [device id]_

- I can see Vector's stimulation and use the customization features in the GUI of wire-pod.

- Vector doesn't seem to be capable of finding his own charger or cube. 

Previously the Vector was set up on:

       Linux Mint Cinnamon (most recent build)
       Older Mac All in One Desktop
  
I assumed the issues were due to the older hardware so I tried this:

       Windows 10 Pro
       Dell Precision Workstation (32 GBs of RAM, Xeon CPU)

This is a production Vector robot. I believe I've been following the correct set-up instructions. 

I'm looking for what steps to try on the present Windows set-up. 

Is this some sort of router settings issue? Internet issue? Houndify issue?

Any help is greatly appreciated. I'm incredibly impressed with this project and am really excited to bring my Vector back to life.

ssmodk, 2024-09-29:
I have this same problem. Vector basically won't do anything despite the logs showing that all the intents are matched.

kercre123, 2026-11-15:
stale; please open a new issue if this is still a concern

closed 2026-11-15 by kercre123