kercre123/wire-pod #264: Allow Alexa only when charging

Zomboy4313, 2024-01-11:
For those people who want to maximise Vector's battery life. But also want more features, I was wondering if it would be possible to only enable the Alexa services when vector is charging.

kercre123, 2024-01-12:
This is embedded in Vector's firmware and the functionality cannot be changed. Alexa takes up negligible resources anyway - just listening for one extra wake word.

closed 2024-01-12 by kercre123

Zomboy4313, 2024-01-12:
I thought Alexa drained battery, or did they fix that?

kercre123, 2024-01-12:
I've never seen any evidence of that. I see no reason why it would

Zomboy4313, 2024-01-12:
Oh, maybe I'm getting information wrong, I recall on some threads that adding Alexa meant taking a hit to battery life due to Alexa's services being power hungry. And Alexa's code itself took almost as much storage as vector's OS. Maybe I'm getting confused.

Anyway, I must give my praise to you, I've bought a Vector recently since I've been interested in one for a while and Wire-Pod's been amazing in keeping Vector alive. Thank you for all of your hard work

Zomboy4313, 2024-01-13:
I found it in https://randym32.github.io/Vector-TRM.pdf, at page 28

kercre123, 2024-01-13:
I personally haven't noticed any difference in battery life or performance with/without Alexa. And it seems to only imply that usage of Alexa may affect battery life, not the activation status, which I don't think is true either.

> "Alexa’s software resources consume as much space as Vector’s main software."

This isn't true.. the the 1.1.0 and 1.2.0 OTAs would be at least 1.5x the size if this were the case, but 1.2.0 is only a few MB bigger.

> “[Alexa Voice Service] solutions for Alexa Built-in products required expensive 
application processor-based devices with 50MB memory running on Linux or Android”

The AVS SDK doesn't take up very many resources. Vector has 512MB of RAM, way more than what is needed. Vector is also far more powerful (and therefore has much more headroom) than the older Echo Dots out there. All it is doing is listening for a wake word (which Vector is already doing), then sending data to/playing back audio from Amazon's servers.

Zomboy4313, 2024-01-13:
Ok, sorry about that. Thank you for clarifying tho