kercre123/wire-pod #265: [HELP] custom intent - custom python script

Wolly2201, 2024-01-13:
First of all, applause for the work you have done. 
Setting up the wire-pod was easy to do with the documentation.
The Wire pod works fine and 'Hey Vector' works well  (change of language was the trick). 
Only I can't manage to execute the custom intent with a script. 

I'm running wire-pod:
  - Debian Bookworm
  - Raspberry Pi 4b
  - Regular prod bot

Prcess staps i've done on raspberry pi with wire-pod. 
- install python3.8
- install cyb3rvector sdk under python3.8
- authenticate with wire-pod (succesfull)
- change example sdk "hello escapepod" to my situation (Hello.py)
- run in terminal  /usr/local/bin/python3.8 ./script/Vector/Hello.py -> works well
- make  a custom intent: 
          Name/Description: Hello
          Utterances: Test
          Intent: intent_greeting_hello
          Exec::  /usr/local/bin/python3.8
          Exec args:  ./script/Vector/Hello.py  (or /home/pi/script/Vector/Hello.py)
-  Vector responds to Test (Intent matched)
-  He does the intent_greeting_hello
-  But not run script Hello.py

Tried a also example 09_eye_color. 
Works fine in terminal but not with custom intent:
Tried also full path of exec args, also with the options of serial bot (example lights-off)

Does anyone know what I am doing wrong?

kercre123, 2024-01-14:
Where exactly is the `script` folder located? wire-pod's working directory is ~/wire-pod/chipper, so if `script` is located in ~/script, you should use `../../script/Vector/Hello.py` in Exec args.

Also note that wire-pod is running as root, so you should also test `sudo /usr/local/bin/python3.8 ./script/Vector/Hello.py`

Wolly2201, 2024-01-14:
Thank you.
The solution was to place the script in the folder chipper. 
Script Hello.py runs now after command Hey Vector!Test.

closed 2026-11-15 by kercre123