kercre123/wire-pod #267: Need account for SDK

Zomboy4313, 2024-01-14:
I want to use the SDK but I need to input Anki Account credentials, and I believe I don't have an account. so what am i meant to do, get someone else's account?

kercre123, 2024-01-14:
Describe how wire-pod is set up. What is it running on? What device are you trying to set the SDK up on? What SDK are you using (there are multiple)?

Zomboy4313, 2024-01-14:
I'm using the native Windows version of Wire-Pod, and I'm trying to install Cyb3rdog's version of the SDK with python 3.8

anjok, 2024-02-27:
I'm trying to use the python SDK on Raspberry pi and have the same problem:

```(.vector) akrank@escapepod:~ $ PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python python -m anki_vector.configure -e foo@bar.com -i 192.168.0.123 -n Vector-U4B3 --s 0dd1b489 

***Anki Vector Python SDK Setup***

Vector requires all requests be authorized by an authenticated Anki user.

This script will enable this device to authenticate with your Vector
robot for use with a Vector Python SDK program.

Vector must be powered on and connected on the same network as your
computer. By running this script, you will be asked to provide your
Anki account credentials, and the script will download an authentication
token and cert that will grant you access to the robot and his
capabilities (such as camera and audio) as well as data stored on the
robot (such as faces and photos).

See the README for more information.

Use of Vector and the Vector SDK is subject to Anki's Privacy Policy and Terms and Conditions.

https://www.anki.com/en-us/company/privacy
https://www.anki.com/en-us/company/terms-and-conditions

Do you wish to proceed? (y/n) y
Using robot name: Vector-U4B3
Using IP: 192.168.0.123
Using robot serial number: 0dd1b489

Downloading Vector certificate from Anki...Traceback (most recent call last):
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connection.py", line 198, in _new_conn
    sock = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/util/connection.py", line 60, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 962, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
socket.gaierror: [Errno -2] Name or service not known

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connectionpool.py", line 793, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connectionpool.py", line 491, in _make_request
    raise new_e
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connectionpool.py", line 467, in _make_request
    self._validate_conn(conn)
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connectionpool.py", line 1099, in _validate_conn
    conn.connect()
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connection.py", line 616, in connect
    self.sock = sock = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connection.py", line 205, in _new_conn
    raise NameResolutionError(self.host, self, e) from e
urllib3.exceptions.NameResolutionError: <urllib3.connection.HTTPSConnection object at 0x7f8924ef50>: Failed to resolve 'session-certs.token.global.anki-services.com' ([Errno -2] Name or service not known)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/akrank/.vector/lib/python3.11/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/urllib3/connectionpool.py", line 847, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
[... 13 lines cut ...]
    cert, serial = get_cert(args.serial)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/anki_vector/configure/__main__.py", line 101, in get_cert
    r = requests.get('https://session-certs.token.global.anki-services.com/vic/{}'.format(serial))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akrank/.vector/lib/python3.11/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='session-certs.token.global.anki-services.com', port=443): Max retries exceeded with url: /vic/0dd1b489 (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7f8924ef50>: Failed to resolve 'session-certs.token.global.anki-services.com' ([Errno -2] Name or service not known)"))
```

There's some hack here, but I couldn't get at the  `scp ...` host:

https://randym32.github.io/Anki.Vector.Documentation/how-to/How%20to%20re-authenticate%20SDK%20apps.html

Zomboy4313, 2024-08-01:
I should mention all errors I had were fixed by downloading [WirePod Vector SDK](https://github.com/kercre123/wirepod-vector-python-sdk)

closed 2024-08-01 by Zomboy4313