kercre123/wire-pod #268: [HELP] Won't recognize my words after "Hey Vector"

Hitojoy, 2024-01-16:
**Describe the issue:**
I followed the instructions and set up the server on my MacBook. it works well, I can control my Vector remotely. But it doesn't respond to whatever I said after "Hey Vector". And then it responds no Wi-Fi (his eyes changes, as if he's in search of signal).

There's an additional information I should add: I'm in China, so vector cannot connect to international network. But under the same subnet, my laptop, where the server is, has access to international network.

**What are you running wire-pod on?**
  - [OS]: Mac OS 14.1.2 (23B92)
  - [Hardware]: MacBook Pro(M2 Pro)

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
Prod bot
**What steps have you tried?**
Reboot vector
Reboot/Reset wire-pod

(note: I will delete any issues which fit under this category but don't follow this template. I will not accept an "It's just not working!")

KnightMars, 2024-07-20:
Have you resolved this issue?Running on macOS, Wirepod shuts down every time a command is spoken!

aofeng188, 2024-11-04:
>你解决这个问题了吗？在macOS上运行，Wirepod每次说出命令时都会关闭！>你解决这个问题了吗？在macOS上运行，Wirepod每次说出命令时都会关闭！Me Too!

kercre123, 2026-11-15:
stale; please open a new issue if this is still a concern

closed 2026-11-15 by kercre123