kercre123/wire-pod #269: [HELP]

xanathon, 2024-01-16:
I reinstalled Wire-Pod, earlier version ran without any problems.

**What are you running wire-pod on?**
  - [OS]: Ubuntu 20.04 LTS
  - [Hardware]: NUC Otazak Ubuntu Mini PC 8GB LPDDR4 128GB SSD Celeron J4125

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**

Unrelated to bot

**What steps have you tried?**

I removed the wire-pod folder and cloned again from git. I ran setup according to the installation instructions in the wiki (with option vosk). When I try to start Wire-Pod I get the following message:

```
# google.golang.org/grpc
/root/go/pkg/mod/google.golang.org/grpc@v1.60.0/server.go:2161:14: undefined: atomic.Int64
note: module requires Go 1.19
```

Earlier version of Wire-Pod ran without this error.

I already tried to purge golang and reinstalled from scratch, but the same error.

TIA!

kercre123, 2024-01-16:
The latest version of wire-pod installs Go 1.19. It used to install 1.18, which is probably what is installed on your system, and wire-pod will not overwrite a Go install (even if it is older than what is required).

How did you try to purge it? It isn't installed through a package manager. wire-pod does it manually.

To purge the install:
```
sudo -s
rm -f /usr/bin/go
rm -rf /usr/local/go/bin/go
```

Then re-run wire-pod's setup script.

xanathon, 2024-01-17:
Addition: well, after a restart it seems to work ...

closed 2024-01-17 by xanathon