kercre123/wire-pod #272: [HELP]

IAmDazen, 2024-01-21:
**Describe the issue:**
when i say "hey vector!" [question] vector either shrugs the question off or says there isnt wifi
**What are you running wire-pod on?**
  - [OS]: Debian Bookworm AKA Raspbian
  - [Hardware]: Raspberry Pi

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
normal, can you dev unlock a wirepod bot?
**What steps have you tried?**
redoing it on pi itself
(note: I will delete any issues which fit under this category but don't follow this template. I will not accept an "It's just not working!")

IAmDazen, 2024-01-21:
you know what... im going to dev unlock it, maybe that will fix it

kercre123, 2026-11-15:
stale

closed 2026-11-15 by kercre123