kercre123/wire-pod #273: [HELP] Vector "What's the Weather" will not work

TypicalZedF, 2024-01-21:
**Describe the issue:**
When I ask vector the weather, he shrugs. I have tried multiple API Keys with openweathermaps, and it does not work. Maybe im doing something wrong? I created the account and verified and it gave me a free api key.
**What are you running wire-pod on?**
  - [OS]: Windows 11)
  - [Hardware]: Desktop Computer, 16GB RAM, Ryzen 5 3500 6-core, Nvidia GTX 1650 Super

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
Regular

**What steps have you tried?**
Generating a new API Key, and waited a couple of hours, and still nothing. Not sure what else to do.

kercre123, 2024-01-21:
What did you set Vector's location to? This usually has something to do with the geolocation API.

TypicalZedF, 2024-01-21:
I set it to my `MyTownName, MyState` in that order.

TypicalZedF, 2024-01-21:
Seems to be a hit or miss for some reason
```
2024.01.21 15:03:44: Intent matched: intent_weather_extend, transcribed text: 'what's the weather', device: 006089e6
2024.01.21 15:03:44: Parameters sent: map[condition:Cloudy is_forecast:false local_datetime:Sunday, 21-Jan-24 15:03:28 EST speakable_location_string:TownNameHere temperature:86.000000 temperature_unit:F]
2024.01.21 15:04:36: Intent matched: intent_weather_extend, transcribed text: 'what's the weather', device: 006089e6
2024.01.21 15:04:36: Parameters sent: map[condition:undefined is_forecast:false local_datetime:test speakable_location_string: temperature:120 temperature_unit:C]
2024.01.21 15:04:48: Intent matched: intent_weather_extend, transcribed text: 'what's the weather', device: 006089e6
2024.01.21 15:04:48: Parameters sent: map[condition:undefined is_forecast:false local_datetime:test speakable_location_string: temperature:120 temperature_unit:C]
```

kercre123, 2024-01-24:
Try setting the location then restarting wire-pod. This seems to have temporarily solved it for me

TypicalZedF, 2024-01-24:
I'll Try that! Thanks! I'll send an update if this works

cl0ne, 2024-01-27:
> Try setting the location then restarting wire-pod. This seems to have temporarily solved it for me

Seems to work, thanks!

TypicalZedF, 2024-01-27:
It ended up working for me. Thought sometimes it does still bug out and refuses to tell me the weather. This is normally fixed by restarting Vector and wire-pod.

kercre123, 2024-01-30:
This is now fixed in the latest commit. The bot would sometimes not return the right Jdoc - causing wire-pod to replace Vector's "RobotSettings" with his "LifetimeStats", which has different information

closed 2024-01-30 by kercre123

TypicalZedF, 2024-01-30:
Noted. I should wait though before updating correct?

TypicalZedF, 2024-02-02:
I was doing some more digging and there is something wrong with using OpenWeatherMap API altogether. The format recommendation I did before was fixed but is still not giving the correct weather information. Going to the OpenWeatherMap website and searching my town, it shows up with Multiple results with no way to correctly tell apart. This is supper confusing and ultimately confusing