kercre123/wire-pod #279: Change STT

MimoAW, 2024-01-28:
I want to change my STT from VOSK to Whisper I can't find any way to do it I have searched through all of the files and in the setup.sh file that is something about STT I couldn't find a way to change it. There's also a vosk file but that doesn't help me. I am using a raspberry pi 4 b does anyone know how to do it? I don't want to set up why a pod all over again 

Thank You

kercre123, 2024-01-28:
Just run `sudo ./setup.sh` and choose Whisper once the prompt comes up. 

I do not recommend using Whisper on a Pi. It doesn't run well

MimoAW, 2024-01-28:
Ok I'll choose a different STT I saw it under the experimental folder 

Thank you for your help!

closed 2024-01-28 by MimoAW