kercre123/wire-pod #280: [BUG]

TypicalZedF, 2024-01-30:
**Describe the bug**
When asking Vector to change his volume, he consistently turns the volume down regardless of the command. If you ask him to turn it up, he will turn it down, if you ask him to turn it down, he turns it down. The only way to change the volume is to set it in the wire-pod bot settings.

**To Reproduce**
Steps to reproduce the behavior:
Ask Vector to turn his volume up or to set his volume to max.

**Expected behavior**
Volume should go up

**Device you are running wire-pod on (please complete the following information):**
 - OS: Windows 11 Pro
 - Hardware: Nivida 1650 Super, Ryzen 5 3500, 16GB RAM DDR4

kercre123, 2024-02-03:
What exactly are you saying to Vector when you are asking to change the volume? I do it by saying "Hey Vector, set your volume to medium high", etc

TypicalZedF, 2024-02-03:
I would say "Hey Vector, Set Volume to Medium/Low/Etc"

kercre123, 2024-05-28:
Are you still experiencing this with the latest version of wire-pod? Could you send logs (accessible in the first page of the web interface) after saying a volume command? It is likely transcribing incorrectly

TypicalZedF, 2024-05-28:
I'm struggling to get it installed on my Linux server..I ended up switching from Windows to Linux for a number of claudi of life and stability reasons. I need to be able to put it in a docker, and I can't have it running on port 80

kercre123, 2026-11-15:
stale

closed 2026-11-15 by kercre123