kercre123/wire-pod #282: Docker installation added to wiki

Zippy-boy, 2024-01-30:
Iv got the markdown for the docker install for the wiki. Apparently you cant create a pull request for the wiki /:

## Guide 5: Docker compose

1. On the device you wo`uld like to install wire-pod on, make sure you have docker installed. This can be done with the command `sudo apt install docker-ce` for linux.
2. Verify if the docker engine is working with the command `sudo service docker status`
3. Run this command to create and download the wire-pod application

```
docker compose up -d -f https://raw.githubusercontent.com/kercre123/wire-pod/main/compose.yaml
```

4. With a device on the same network as wire-pod, open a browser and head to the configuration page. <http://YOUR_IP:8080>. In that page, follow the instructions. Wire-pod should then be set up!
5. Continue on to ["Authenticate the bot with wire-pod"](#authenticate-the-bot-with-wire-pod), near the bottom of this page.

g0t4, 2024-05-11:
IIAC you are using `linux/amd64`?

Zippy-boy, 2024-05-12:
I am using linux/amd64 on my server.

TypicalZedF, 2024-05-12:
Is this just a suggestion, or does this work?

Zippy-boy, 2024-05-21:
This does work for an Ubuntu install. And is very similar for other OS's running docker.

TypicalZedF, 2024-06-15:
What if i can't use port 8080...? Every software wants to use port 80 or 8080-

Zippy-boy, 2024-06-15:
Then you can curl the compose file and change the port to something else. And then compose up

TypicalZedF, 2024-06-15:
Do you think I could get this to work with Casa OS?

Zippy-boy, 2024-06-15:
Yep, 100% Just make sure that when you change the port for the web UI change the left port but keep the right one at 8080. Like this 2342:8080

TypicalZedF, 2024-06-15:
Installing energetic jorja Error Info

Error response from daemon: manifest unknown

Zippy-boy, 2024-06-15:
Command still seems to work for me. Try `curl -O https://raw.githubusercontent.com/kercre123/wire-pod/main/compose.yaml`
And then
`docker compose up -d`

TypicalZedF, 2024-06-15:
Something is wrong with the compose. Try running `docker search ghcr.io/kercre123/wire-pod` on my terminal to check and well...
```
permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get "http://%2Fvar%2Frun%2Fdocker.sock/v1.45/images/search?term=ghcr.io%2Fkercre123%2Fwire-pod": dial unix /var/run/docker.sock: connect: permission denied
```

Zippy-boy, 2024-06-15:
Huh, odd. The image should be available to the public as it's a public repo.
Try `sudo usermod -aG docker ${USER}`
And `sudo chmod 666 /var/run/docker.sock`

And then restart the computer.

If this doesn't work, you may have to docker login with github, but I don't think that's the issue.

TypicalZedF, 2024-06-15:
This didn't seem to work

Zippy-boy, 2024-06-15:
Now that's odd. Il have a look into that later. For now, it might be easier to build the docker image yourself then. Give me a sec and il get you the commands (:

Zippy-boy, 2024-06-15:
So, pull the repo `git clone https://github.com/kercre123/wire-pod.git`
`cd wire-pod`
Replace the compose.yml file with 
`version: '3.8'
services:
  wire-pod:
    hostname: escapepod
    build: 
      context: .
    restart: unless-stopped
    ports:
      - 443:443
      - 8080:8080
      - 80:80
      - 8084:8084
    volumes:
      - wire-pod-data:/chipper/
      - wire-pod-model:/vosk/
volumes:
  wire-pod-data:
    driver: local
  wire-pod-model:
    driver: local`

And then `docker compose up -d`

YoussefElSayad, 2024-07-04:
@Zippy-boy Hello there, I received the error below while attempting to docker compose up on a raspberry pi, is there a way around it?

Error: "no matching manifest for linux/arm64/v8 in the manifest list entries"

Thank you for this!

Zippy-boy, 2024-07-04:
Hey, I haven't tried running docker wirepod on a raspberry pi yet, but I think if you add `platform: linux/amd64` inside the wirepod service in the `compose.yml` file it should work. Let me know if you need any more help :)

YoussefElSayad, 2024-07-04:
Thank you for the quick response, that did work! Now I just gotta figure out what ports to use instead of 80, 443, 8080 and 8084.

Thank you!

Zippy-boy, 2024-07-04:
@YoussefElSayad, the only port you can change is 8080, I'm pretty sure. The vector firmware forces the rest. Don't change them!!

YoussefElSayad, 2024-07-04:
@Zippy-boy What can i do if these ports are being used though? I think they are currently used by my current wire-pod installation on the raspberry pi but I wanted to ensure docker was running correctly in order to keep my vector up and running

Zippy-boy, 2024-07-04:
I would recommend stopping the wirepod service and testing with the docker version. If you find that it doesn't work, you can always stop the docker version and start back up the local one, no files or configs will be lost

YoussefElSayad, 2024-07-24:
Hello, it's me again! I managed to locally build an arm64 image to suit my needs and that works great, however it seems that my volumes are not mapped correctly, as it redirected me to the initial setup screen. I currently have the repo cloned in a folder labeled wire-pod, and i checked the docker container logs, which provided me with this statement:

```
It looks like you may have changed path names of the directories wire-pod is running in. This is not recommended because the SDK implementation depends on relativity in a few spots.
Unsupported path scenario, printing podPathSplit: 
[ chipper]
```

If anyone has experienced this or knows a way around it to load my local volumes, I'd appreciate it!

YoussefElSayad, 2024-07-24:
Hello, it's me again! Just thought I'd update this thread with how I managed to get it to run on my raspberry pi

By simply using docker buildx i was able to create a new image locally with my preferred architecture, and afterwards i updated the docker compose file to use the newly created image, and tada! At first it brought me to the initial setup screen, however i later managed to have it read my existing data.

I'd be more than happy to help if anyone faced the same issue i faced, and thanks again @Zippy-boy for the prompt and helpful responses!

jbumgit, 2024-11-08:
@YoussefElSayad Hi. Which raspberry pi did you build your image for?  I currently have a Pi 4b and using Pi OS (bookworm, 32bit with arm_64bit=0 -> due to some other issues going 64bit).  I have been trying to get docker compose working but kept running into some issues at the STT=vosk ./setup.sh step of the docker build due to a user prompt for STT selection.  Since 32bit doesn't have vosk support, picovoice needs to be specified.

My only workaround was to update setup.sh and set some values related to STT and picovoice to get the script through without user intervention.

Were you able to complete the build with the script as is?

jbumgit, 2024-11-08:
I proceeded to build the docker image by manually editing a couple areas in the setup.sh script to get through the STT selection logic (chose picovoice).

The docker image starts up fine and I can hit the setup page.  It's now stuck at "Initiating setup..." after clicking on "submit".  How can I troubleshoot what's going on?  docker logs doesn't show anything beyond "Configuration page: http:/..."

Thanks.