kercre123/wire-pod #285: [HELP]  escapepod.local works, but vector does not activate

Sloth-on-meth, 2024-02-02:
**Describe the issue:**

**What are you running wire-pod on?**
  - [OS]: Ubuntu server
  - [Hardware]:  not relevant, generic server

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
Prod

**What steps have you tried?**
When i tryi to activate, i get 

"Error logging in,. the bot is likely unable to communicate with your wirepod [....]'

i setup my local dns server so escapepod.local points to 192,168,178,79 , which is where my server is running, and it _does_ work when trying in the browser

Sloth-on-meth, 2024-02-05:
https://github.com/kercre123/wire-pod/files/14164925/vector-logs-2024-02-05-12-29-52.tar.bz2.txt

Rodziac, 2024-02-29:
Hello! I'ts been 3 weeks, so you maybe fixed it or gave up already, but:

1. It seems you are saving your log file wrong. It's actually a `tar.bz2` file, but since you have it saved as `tar.bz2.txt` it gets opened like a text file. This is actually an archive file. You need to:
  i. Rename the file to remove `.txt` at the end
  ii. Open it with your file archiver (I use 7-zip, but Windows nowadays is also able to open this archive file)
2. Now that you can open the log archive, you can look inside.
  i. If you open `data/diagnostics/logs/log.txt` in your text editor, you can see the actual logs!
  ii. If you read around the end, you'll see it says: `"transport: Error while dialing dial tcp: lookup escapepod.local: no such host"`

It seems like your vector is not able to reach the escapepod.local domain. Your Vector maybe is not using your local DNS server. I wonder if your router is set up to use/advertise your local DNS server.

There are log files that show vector ping anki.com. I wonder if you could update your local DNS server to point anki.com to some other IP, restart the vector, check the logs, and see which IP address it pings. If it pings the IP address you set, it means vector is indeed using your local DNS. If not, it means your vector is not using your local DNS so you'll have to look into that.

There -could- be other reasons why it works on your browser but not on vector, but this is the simplest way you can start troubleshooting...

kercre123, 2024-03-02:
wire-pod is a hobby project, not something I can devote all of my time to, so I can't respond to all issues in a timely manner.

Have you tried the [troubleshooting steps in the wiki](https://github.com/kercre123/wire-pod/wiki/Troubleshooting#error-logging-in-the-bot-is-likely-unable-to-communicate-with-your-wire-pod-instance)? Maybe try unsetting the rule in your DNS server - wire-pod does mDNS broadcasts which should work if your network allows multicast.

kroteau, 2024-05-23:
I've had quite a time debugging an issue just like this.
Your router might block mDNS packets(for example mikrotik does this, guest network on a router also block any interaction between connected hosts).
Just try another router!

kenny-nt, 2024-05-27:
Just wonder why wire-pod don't let us change to a custom host? If escapepod.local can not be resolved in activation, just show a input for entering the wire-pod custom host...

webguy1dev-dc, 2025-01-20:
His file is send is: https://drive.google.com/drive/folders/1SVxqzUZP7ANaD-k3FTSQD4EkUloznsdD?usp=drive_link

closed 2026-11-15 by kercre123