kercre123/wire-pod #286: [BUG] "Hey Vector, set a timer for X Hour(s)" does not work as expected.

TypicalZedF, 2024-02-02:
**Describe the bug**
When asking Vector to set a timer with hours he does not understand the command correctly as shown in the logs bellow:

User: "Hey Vector, Set a **2 Hour** Timer"
Response: * Sets a 2-second timer * - This response is incorrect.
```
2024.02.02 17:33:39: Intent matched: intent_clock_settimer_extend, transcribed text: 'set a two hour timer', device: 006089e6
2024.02.02 17:33:39: Parameters sent: map[timer_duration:2]
```

User: "Hey Vector, Set a **10 Minute** timer
Response: * Sets a 10 Minute Timer * - This is the correct response
```
2024.02.02 17:37:12: Intent matched: intent_clock_settimer_extend, transcribed text: 'set a timer for ten minutes', device: 006089e6
2024.02.02 17:37:12: Parameters sent: map[timer_duration:600]
```

**To Reproduce**
Simply ask Vector to set a timer specifying the number of minutes. This behavior is correct. Then ask him to set an hour-based timer, he does not seem to understand the task very well. 

**Expected behavior**
What should happen is vector should understand the difference between "Minute" and "Hour", if the word hour is said this should simply note the number of hours followed by just an additional 60 minutes. Due to Vector's Display, he should be limited to 99 Minutes or 1.65 Hours. If possible, I also think it would be great for Vector to use more than a limited amount of hours by falling back to a `Hours : Minutes` format, then fallback to `Minutes : Seconds` once the Hours value is set to 0.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device you are running wire-pod on (please complete the following information):**
 - OS: Windows 11 Pro
 - Hardware: Ryzen 5 3500 6-Core, 16GB DDR4 RAM, Nividia GTX 1650 Super

kercre123, 2024-02-03:
I believe the maximum is 1 hour - set in Vector's firmware which cannot changed.

kercre123, 2024-05-28:
The timer code has been updated by this point and should work when "one hour" is said.

closed 2024-05-28 by kercre123