kercre123/wire-pod #287: [HELP] Is it possible to use custom STT Models?

TypicalZedF, 2024-02-02:
**Describe the issue:**
I was simply wondering if it was possible to use custom STT models. I know it would cost more on performance but I notice services like Alexa and Google have better STT models which allow them to understand speech better. I was hoping I could use a custom model with vector so he could understand me a whole lot better. Aka it would allow me to speak to him like a person, and less like a robot (Pausing in the middle of words so he understands me clearly)

**What are you running wire-pod on?**
  - [OS]: Windows 11 Pro
  - [Hardware]: Ryzen 5 3500 6-Core, 16GB Ram, Nividia 1650 Super

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
Regular bot

TypicalZedF, 2024-02-02:
Sorry I noticed this already had a closed issue report for this:
https://github.com/kercre123/wire-pod/issues/279

kercre123, 2026-11-15:
stale

closed 2026-11-15 by kercre123