kercre123/wire-pod #288: [HELP] robot sdk guid changes every 24 hours

a7ypically, 2024-02-07:
I'm using the python SDK for a long running program and it fails every 24 hours due to the guid changing for the robot. The bigger issue is that it leaves the robot in a bad state - for example no more images coming from the robot until a restart.
I wonder if this 24 hours refresh is forced by the robot or something configurable on wire-pod?

kercre123, 2024-03-16:
wire-pod could be modified so the JWT token expires after a month or so rather than a day - i'll do that in the next commit. Also, the robot does a maintainence reboot every 24 hours which may be breaking your connection. The SDK should be getting the new GUID from the bot with the (unchanging) session certificate each time the script connects to the robot

a7ypically, 2024-03-17:
I just hacked my installation of wire-pod to just return the same token and it's fine. I did not know about the 24 hours reboot. However I expect the same token to still be valid on the robot as it is stored on disk.

kercre123, 2026-11-15:
stale

closed 2026-11-15 by kercre123