kercre123/wire-pod #290: [BUG] SayText -> Internal Error: Failed to say Text at Dashboard and via Python SDK

dasTholo, 2024-02-28:
**Describe the bug**
I have installed the Wire Pod via Docker on my Arch Linux desktop. The bot configuration works (eye color, move, camera tested)

However, I am not able to convince Vector to talk. Neither via the Wire-Pod interface (in Control Center I have activated Behavior Control) nor via a Python script. Therefore I have to assume that the GRPC commands have a bug.
I would like to explain this using the following example.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to: Bot Settings -> Vector Control
2. Click on : Behavior Control -> Assume
3. Scroll down to 'say Text' -> Submit
4. See error: "No Error in Docker Logs

**Device you are running wire-pod on :**
 - OS: 
 ~ docker version
Client:
 Version:           25.0.2
 API version:       1.44
 Go version:        go1.21.6
 Git commit:        29cf629222
 Built:             Thu Feb  1 10:50:44 2024
 OS/Arch:           linux/amd64
 Context:           default

**Python dependencies:**
> dependencies = [
>     "ikkez-vector @ git+https://github.com/ikkez/vector-python-sdk@a5f994fbecb3729687d690513d98efd319947847",
>     "zeroconf>=0.131.0",
> ]

Also tested with old protobuf

> dependencies = [
>     "zeroconf>=0.131.0",
>     "kingardor-vector @ git+https://github.com/kingardor/vector-python-sdk@d6dfac72ad4a044771913c99b24ce37255f24a17",
>     "protobuf==3.20.0",
> ]

**Working test script:**
```
import anki_vector

def main():
    with anki_vector.Robot(
        serial="xxxxx",
        name="Vector-xxxx",
        ip="192.168.178.46",
        config={"cert": "/home/tholo/PycharmProjects/vector/config/Vector-xxx-xxxxxx.cert", "name": "Vector-xxxx", "guid": "xxxxx=="},
        behavior_activation_timeout=30.0,
        cache_animation_lists=False,
        default_logging=True
    ) as robot:
        print(robot.get_battery_state())
```

**Response:**

> /home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/anki_vector/mdns.py:75: FutureWarning: <anki_vector.mdns._MdnsListener object at 0x7bc6911007d0> has no update_service method. Provide one (it can be empty if you don't care about the updates), it'll become mandatory.
>   ServiceBrowser(zeroconf, "_ankivector._tcp.local.", listener)
> 17:06:47.302     connection.Connection    INFO  Connecting to 192.168.178.46:443 for Vector-xxx using /home/tholo/PycharmProjects/vector/config/Vector-xxxx-xxxxx.cert
> 17:06:47.794     connection.Connection    INFO  BehaviorControl control_granted_response { }
> 17:06:47.794     events.EventHandler WARNING  Unknown Event type
> battery_level: BATTERY_LEVEL_NOMINAL
> battery_volts: 4.0167966

**SayText with Error Sript:**
```
import anki_vector

def main():
    with anki_vector.Robot(
        serial="xxxxx",
        name="Vector-xxxx",
        ip="192.168.178.46",
        config={"cert": "/home/tholo/PycharmProjects/vector/config/Vector-xxxx-xxxxxx.cert", "name": "Vector-xxxx", "guid": "xxxxx=="},
        behavior_activation_timeout=30.0,
        cache_animation_lists=False,
        default_logging=True
    ) as robot:
    robot.behavior.say_text("Hello World")
```
**Response:**

> /home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/anki_vector/mdns.py:75: FutureWarning: <anki_vector.mdns._MdnsListener object at 0x743cc33887d0> has no update_service method. Provide one (it can be empty if you don't care about the updates), it'll become mandatory.
>   ServiceBrowser(zeroconf, "_ankivector._tcp.local.", listener)
> 17:22:29.515     connection.Connection    INFO  Connecting to 192.168.178.46:443 for Vector-xxx using /home/tholo/PycharmProjects/vector/config/Vector-xxx-xxxx.cert
> 17:22:29.960     connection.Connection    INFO  BehaviorControl control_granted_response { }
> 17:22:38.062       events.EventHandler WARNING  Unknown Event type
> Traceback (most recent call last):
>   File "/home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/anki_vector/connection.py", line 753, in log_handler
>     result = await func(*args, **kwargs)
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/anki_vector/behavior.py", line 205, in say_text
>     return await self.conn.grpc_interface.SayText(say_text_request)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/aiogrpc/channel.py", line 40, in __call__
>     return await fut
>            ^^^^^^^^^
> grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
> 	status = StatusCode.INTERNAL
> 	details = "Failed to say text"
> 	debug_error_string = "UNKNOWN:Error received from peer ipv4:192.168.178.46:443 {grpc_message:"Failed to say text", grpc_status:13, created_time:"2024-02-28T17:22:38.195848595+01:00"}"
> >
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/home/tholo/PycharmProjects/vector/src/vector/test.py", line 40, in <module>
>     main()
>   File "/home/tholo/PycharmProjects/vector/src/vector/test.py", line 18, in main
>     robot.behavior.say_text("Hello World")
>   File "/home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/anki_vector/connection.py", line 811, in result
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "/home/tholo/.rye/py/cpython@3.12.1/install/lib/python3.12/concurrent/futures/_base.py", line 456, in result
>     return self.__get_result()
>            ^^^^^^^^^^^^^^^^^^^
>   File "/home/tholo/.rye/py/cpython@3.12.1/install/lib/python3.12/concurrent/futures/_base.py", line 401, in __get_result
>     raise self._exception
>   File "/home/tholo/PycharmProjects/vector/.venv/lib/python3.12/site-packages/anki_vector/connection.py", line 756, in log_handler
>     raise connection_error(rpc_error) from rpc_error
> anki_vector.exceptions.VectorConnectionException: StatusCode.INTERNAL: Failed to say text
> 
> Unknown error
> Process finished with exit code 1

Can you help me with this Error?
How can i get at deeper logs in Wire-Pod? Doesn't found any variable or anything.

I've seen this Proto at my python module. Is the Proto the same at wirepod? Sry i can´t understand go stuff
```
 // Make Vector speak text.
  rpc SayText(SayTextRequest) returns (SayTextResponse) {
    option (google.api.http) = {
      post: "/v1/say_text",
      body: "*"
    };
  }
```

Thanks for your help and hard work!!

frien137, 2024-02-29:
python 12 with ikkez sdk  using pycharm and docker

kercre123, 2024-03-02:
Closing this as it has been figured out on the Discord.

The locale was set to de-DE, and there is no STT package for that in Vector's firmware, causing an error.

closed 2024-03-02 by kercre123