kercre123/wire-pod #294: [HELP] Issue with running ./chipper/start.sh On Rasberry Pi

apeoplerson, 2024-03-06:
**Describe the issue:**
wirepod@escapepod:~/wire-pod $ sudo ./chipper/start.sh
# command-line-arguments
/usr/local/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
/usr/bin/ld: skipping incompatible /usr/lib/i386-linux-gnu/libopus.so when searching for -lopus
/usr/bin/ld: skipping incompatible /usr/lib/i386-linux-gnu/libopus.a when searching for -lopus
/usr/bin/ld: cannot find -lopus
/usr/bin/ld: skipping incompatible /usr/lib/i386-linux-gnu/libopus.so when searching for -lopus
/usr/bin/ld: skipping incompatible /usr/lib/gcc/i686-linux-gnu/10/../../../libopusfile.so when searching for -lopusfile
/usr/bin/ld: skipping incompatible /usr/lib/gcc/i686-linux-gnu/10/../../../libopusfile.a when searching for -lopusfile
/usr/bin/ld: skipping incompatible /lib/libopusfile.so when searching for -lopusfile
/usr/bin/ld: skipping incompatible /lib/libopusfile.a when searching for -lopusfile
/usr/bin/ld: skipping incompatible /usr/lib/libopusfile.so when searching for -lopusfile
/usr/bin/ld: skipping incompatible /usr/lib/libopusfile.a when searching for -lopusfile
/usr/bin/ld: cannot find -lopusfile
/usr/bin/ld: skipping incompatible /usr/lib/gcc/i686-linux-gnu/10/../../../libopusfile.so when searching for -lopusfile
/usr/bin/ld: skipping incompatible /lib/libopusfile.so when searching for -lopusfile
/usr/bin/ld: skipping incompatible /usr/lib/libopusfile.so when searching for -lopusfile
collect2: error: ld returned 1 exit status

**What are you running wire-pod on?**
  - [OS]: Debian
  - [Hardware]: Rasberry Pi

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
Regular

**What steps have you tried?**
I tried everything. I don't know how to get past this. Something to do with compiling. Anyway help would be great.

frien137, 2024-04-28:
which wire-pod installer did you use?  try  wirepod_arm64-v1.1.18.deb

bootstrap56, 2025-11-13:
Hi vivascau. I'm also having trouble installing wirepod on my Pi via the method outlined in the installation guide. Something related to a 'Cannot find ld' problem that I can't seem to resolve. I'm now interested in the arm deb method as a workaround. Can you please point me to the Guide 1 that you mention?