kercre123/wire-pod #296: [BUG]

jamesfawcett, 2024-03-16:
**Describe the bug**
GPT Prompt not saving on latest update. After refresh or revisiting the page, it says "undefined".

**To Reproduce**
Steps to reproduce the behavior:
Go to :8080/setup.html
Enter a prompt, or the default. 
Save

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Device you are running wire-pod on (please complete the following information):**
 - Hardware: Armbian
 
Console logs:

[image]

updateIntentSelection("editSelect")
updateIntentSelection("deleteSelect")
createIntentSelect("intentAddSelect") <---

kercre123, 2024-03-16:
How exactly did you update it? It seems like the webroot has been updated, but the actual server code hasn't been recompiled. You probably need to rebuild the binary, which can be accomplished with the steps here [https://github.com/kercre123/wire-pod/wiki/Things-to-Know#linux](https://github.com/kercre123/wire-pod/wiki/Things-to-Know#linux)

jamesfawcett, 2024-03-16:
Thank you @kercre123 I'd not seen these other pages of the Wiki - just checking :)

jamesfawcett, 2024-03-16:
@kercre123 thank you - I was only half upgraded! I see the changed custom intents screen now, thank you for all your hard work :)

closed 2024-03-16 by jamesfawcett