kercre123/wire-pod #297: [BUG] invalid memory address or nil pointer dereference

YoungjuneKwon, 2024-03-17:
**Describe the bug**
Hello, I got below error messages, after I had executed chipper/start.sh

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xf3a42d]

goroutine 50 [running]:
github.com/kercre123/zeroconf.(*Resolver).Browse(0x0, {0x14a04e8, 0xc000318180}, {0x134b151, 0x10}, {0x1340c5d, 0x6}, 0xc0003520c0)
        /root/go/pkg/mod/github.com/kercre123/zeroconf@v1.0.1/client.go:93 +0x12d
github.com/kercre123/wire-pod/chipper/pkg/mdnshandler.PostmDNSWhenNewVector()
        /home/shining/documents/wire-pod/chipper/pkg/mdnshandler/mdns.go:33 +0xd4
created by github.com/kercre123/wire-pod/chipper/pkg/mdnshandler.PostmDNS
        /home/shining/documents/wire-pod/chipper/pkg/mdnshandler/mdns.go:69 +0xb8
exit status 2
```

**Environments**
My environment is Windows 11, and I installed Ubuntu 22.04 on WSL1.
I cloned the code today, and I followed the instructions on README.md.

kercre123, 2024-03-17:
Does this happen every time you try to start it? A similar error happened to me but it was one-off and never happened to me again. Any reason you aren't using the precompiled Windows release [https://github.com/kercre123/WirePod/releases/latest](https://github.com/kercre123/WirePod/releases/latest)

YoungjuneKwon, 2024-03-19:
I noticed that my Windows 11 was rejecting mDNS-related UDP connections. I'm still not sure why, so I've been adding exceptional statement to the code around using ZeroConf objects in the chipper and token related code, then I changed the hostname into escapepod to use the system. I will share the reason of the rejection by my Windows if I know why.

closed 2024-03-19 by YoungjuneKwon