kercre123/wire-pod #303: [BUG] Adding Knowledge Graph Setup

Gecko60, 2024-04-04:
**Describe the bug**
When I put in any type of API Provider and the API Key I then go down and click on the Yes radio button and I do not see a Robot Name section, also further more, when I click save literally nothing happens. It does not save. I tried all three different API Providers and nothing saves. When you click on the save button it does nothing. I have tried the web interface I have even tried to manually go in to apiConfig.json and input the information still does not work, when you try to ask Vector a question his respose is only "knowlegde graph is not enabled, this can be enabled in the web interface."

**Device you are running wire-pod on (please complete the following information):**
 - OS: [Windows]

infinitybs, 2024-04-04:
I have a similar if not the same issue on a few install on a PI. OpenAI And Together won't save, but None and Houndify will

kercre123, 2024-04-12:
The latest release should work fine now;

closed 2024-04-12 by kercre123

kercre123, 2024-04-16:
What method did you use for installation and what method are you using to update? My guess is that you did a git pull, but didn't rebuild the chipper binary - so the older code is still running

infinitybs, 2024-04-16:
> I did this to update, and Guide 2 to install
> 
> sudo systemctl stop wire-pod cd ~/wire-pod sudo ./update.sh sudo ./setup.sh daemon-enable sudo systemctl start wire-pod

Got it working by running this again. Thanks for your help

cd ~/wire-pod
sudo STT=vosk ./setup.sh