kercre123/wire-pod #320: Won't tell me the correct weather

MimoAW, 2024-04-26:
I live in the United Kingdom and when I ask Vector about the weather it's always wrong I have it all set up correctly, but I'm not sure how to format my location. It uses a state but the United Kingdom has no states so how would I format mine?

Zomboy4313, 2024-04-26:
I live in the Uk and Vector is fine for me, which weather api are you using?

MimoAW, 2024-04-26:
I am using weather API (not open weather map)

Zomboy4313, 2024-04-26:
If it helps I formatted my location as Town, Country such as Pontypool, Wales

MimoAW, 2024-04-26:
Okay thank you. I will try that

MimoAW, 2024-04-26:
Hi, I just changed my location to Chelmsford, England Which is very close to where I live I asked him what the weather is after restarting wirepod and he said it's 2° in snowing when it's 10° and cloudy

Zomboy4313, 2024-04-26:
Try using OpenWeatherMap, when I used WeatherAPI its readings were off (So maybe it's not as accurate in the UK?). See if OpenWeatherMap has better results because it did for me, plus it has forecast support

Zomboy4313, 2024-04-26:
Glad I could help :)